nevermore3d/Nevermore_Micro #4: Question : Granules vs Pellets?

PhilBaz, 2022-01-11:
Would the micro v4 or v5 work with granules rather than pellets? I could sift of any dust. Do you see a problem with that plan?

eduncan911, 2022-01-12:
I'll let @0ndsk4 chime in.  

Until then, IMO the irregular surface area of granules would mean you could have some blockage from two or more pieces clumping together.  Whereas pellets generally wouldn't clump no matter how they are shifted around.

0ndsk4, 2022-01-12:
TLDR - Granules are good too. 
Pro: Often slighter higher efficiency/surface area due to intact porosity/structure compared to pellets that are extruded with a binding agent. 

Con: more dusty, generally much higher pressure drop. 4x6 mesh are big enough particles to have equal pressure drop compared to 4mm pellets. 4x8 is comparable to 3mm pellets, and generally have a lot poorer air flow with 5015 fans - if using these make sure you have a high static pressure fan and don’t pack it (sprinkle fill the cartridge).  Anything smaller than 4x8 is unusable.

closed 2022-01-12 by eduncan911

PhilBaz, 2022-01-12:
Thanks for the info, very helpful. I would buy nevermore carbon, but alas I live in the frozen northern tundra known as Canada, and shipping ain't cheap!

Two quick questions...

1) roughly how many grams of carbon fill the V4 or V5
2) In the V5 folder, which files am I supposed to use? V5/V2?

0ndsk4, 2022-01-13:
Good carbon is lighter (more porous) so weight is a bad metric. But about 50 grams. 

Naming scheme is for voron machines (ie Voron 2.4, V2, or Voron 1.8, or V0). For general use and placement on a 2020 extrusions machine, V1.8 is most versatile. 

For Canada, the guy running Jurassic Carbon (Toronto) has been vouched for on discord. His 4mm Anthracite pellets is a very good option for printer fumes.

PhilBaz, 2022-01-13:
Thank you!!!

I found a local supplier with 4mm CKV4 carbon pellets....

Just need to look into the CKV4 designation a bit more closely... Im hoping these will work out.