nevermore3d/Nevermore_Micro #9: BOM error - Magnets are '3x6' not '6x3'

RudyAlbertoAcostaMarmalade, 2022-03-23:
As the title states - I now have to find some creative use for some very long and thin cylindrical magnets :D

wkeifenheim, 2022-04-18:
I nearly made the same mistake, but when I couldn't find 3x6 magnets on Amazon I read closer and saw in the v5 changes section states `BOM 3x6 magnets are standard`. Standard BOM is 6x3 so I then compared the v5 plenum to other Voron V2 parts and confirmed that the BOM listing should state 6x3.
[image]

wkeifenheim, 2022-04-18:
Someone has already submitted a PR to fix this: https://github.com/nevermore3d/Nevermore_Micro/pull/5

0ndsk4, 2022-04-25:
Thanks, is corrected in the V5 BOM section now!

closed 2022-04-25 by 0ndsk4