nevermore3d/Nevermore_Micro #11: Artefacts in Nevermore Area

DietZnX, 2022-06-07:
Hello , I print the V2 Duo_Catridge and found some artefacts in the front area on my print . I see that it comes from the stl . Enclosed some pictures . !

I love the Nevermore ! Thank you for that :) 

[image]

[image]

0ndsk4, 2022-07-09:
Strange, I'll investigate that. Thanks.

MandicReally, 2022-09-11:
Came to point out this same thing. Had hoped to just re-export the STL as it looks like a bad export to me, but the CAD files don't seem to include either cartridge housing.

bucknova, 2023-03-29:
Verified artifacts on STL file. 
Will look into clearing up.

bucknova, 2023-04-01:
Inspected F3D files. Artifacts are not on original models. 
I will export and re-upload the STL files.

bucknova, 2023-04-10:
Cause of artifacts determined to be extruded text and fillets. 

Reworked the texts, and filleted to a smaller degree. New design has a cleaner topography and tessellates better in slicing software. 

I will update all remaining pieces. After the parts have been updated for support material, they will be published.

Neolker, 2023-04-12:
Can I just ask for this part, I've been waiting for it for quite a while, or will there be major changes on the other parts as well so I have to wait anyway? Thanks in advance.

bucknova, 2023-04-12:
Files have been updated. 
I will have them pushed to the GitHub shortly. 

Issue has been resolved.

closed 2023-04-12 by 0ndsk4