nevermore3d/Nevermore_Micro #22: Revise V5 BOM Heatset insert locations for plenum

Flipsoul, 2023-01-21:
1x Plenum base to Plenum lid joint
4x for Fan tabs (2 ea fan)
---
Total 5x at Plenum

Also, it seems most assemblies people are making omit the heatset inserts and SHCS that go into the fan tabs, while adding the fastener in cartridge base to lid joint. including referenced assembly pictures for V5.

MarcelRobitaille, 2023-05-07:
Also curious about this. It seems like the numbers don't add up in the readme.

Shouldn't it be 6 total?
- 1x in plenum base to secure plenum lid
- 1x in cartridge base to accept cartridge lid
- 4x for fans

I'd be happy to take pictures for the gallery and make a PR updating the readme if I could get some confirmation that this is correct. 

[image]

0ndsk4, 2023-05-08:
Hi, true, bom is for the original mounting position in a voron v2, where the base insert isn’t used. A clarification that a fifth (sixth in total) insert is needed for panel mounting on a 2020 extrusion is a good idea. Please proceed :)

closed 2023-05-08 by 0ndsk4

0ndsk4, 2023-05-08:
Blind hole shouldn’t be there. Originally I just put it there for symmetry and had a short blind bolt in it. I’ve since gotten used to the fact the useful hole is off center.