nevermore3d/Nevermore_Micro #23: V5 Slide-in Magnet mod interference with lip on lid

Flipsoul, 2023-01-21:
Slide in magnet mod adds slots for magnets to roll-in on at front of plenum. These interfere ~1.5 mm with the lip on the plenum lid. easy to clip the added feature back. But it seems it could be reduced without loss to printability or function

mechatroniks-git, 2023-01-24:
I'm looking at the CAD model, getting ready to print this. I'm not seeing any interference. Can you confirm where you are seeing it?

[image]

Flipsoul, 2023-01-25:
i wont be able to verify or get pictures till the end of the week, but i believe my particular issue was between prints of
Nevermore_Micro/V5_Duo/Mods/Slide-in-Magnets_by-ChainsDDNevermore_Micro/V5_Duo/Mods/Slide-in-Magnets_by-ChainsDD/Duo_Plenum_wago.stl
And
Nevermore_Micro/V5_Duo/V2/V2_Duo_Plenum_LID.stl

There are a quite a few different solid model files/mesh files, so its possible i have a mismatch

Flipsoul, 2023-01-25:
Looks like it was a mismatch. I'm seeing more files within the slide in magnet mod folder than i did before including the plenum lid file, which i want seeing before. (Couldn't see any of the images in the plenum lid file before too) Not sure why i want seeing them before.

closed 2023-01-25 by Flipsoul