nevermore3d/Nevermore_Micro #27: Improvements to the readme for new users

LiSongMWO, 2023-03-16:
I'm a new user and I read the `readme.md` and while it left me more informed on VOCs and carbon in general, it didn't actually help me better understand the build, and how to integrate it into a printer, or what it actually is (i.e. will I be able to use it?). I had to find some random dude on youtube doing a video on installing it (and installing it in a bad (hot) location) before I understood what and how. The shiny renders look nice but they don't really give me much information, I'd much rather have seen an exploded diagram of the main parts and how they connect/attach to a printer.

The biggest thing I missed was an overview of how it's meant to be used and installed. Something like the following at the top of the readme would have helped me greatly: "Nevermore Micro is a recirculating activated carbon air filter installed inside of a sealed print chamber. Typically it's installed by attaching to the printer's t-slot extresions or VHB taped to the chamber walls. It needs one free 12V/24V fan header on your control board and firmware support (e.g. klipper)."

Additionally while there's a lot of good information on the readme, it's very long and a bit unfocused. My suggestion is to move historical context and background on VOCs etc to a separate (github wiki?) page and focus the readme on building, BOM, sourcing and installing linking out to the above pages in the introduction.

And thanks for this cool upgrade :)

agravelot, 2023-06-08:
Here is a documentation which can be useful: https://www.ldomotion.com/p/guide/18295873486193941

jwyse, 2024-02-01:
LDO moved their guides:
[Nevermore V5 Duo - V2.4](https://www.ldomotion.com/p/guide/Nevermore-V5-Duo)
[Nevermore V5 Duo - Trident](https://www.ldomotion.com/p/guide/Nevermore-V5-Duo--Trident)

FJRPilot, 2024-02-11:
I would also chime in and suggest that specific instructions be added regarding recommended print orientations for the V6 Micro.  Importing the STL's bring the Plenum and Cartridge into slicers vertically. If you attempt to slice you will get errors about "Floating Cantilever's" and asked to re-orient the objectsS
[image]

jwyse, 2024-02-12:
> I would also chime in and suggest that specific instructions be added regarding recommended print orientations for the V6 Micro. Importing the STL's bring the Plenum and Cartridge into slicers [...]

The default orientation (vertical) is actually the recommended orientation.  Orca's floating-cantilever warning can be ignored.  A note in the readme to clarify this would be nice.