nevermore3d/Nevermore_Micro #29: nevermore V5_duo - Plenum lid make fans loud and weak!

Hrumque, 2023-04-02:
The project looks very nice optically, but in practice - I had my doubts when printing, and the first launch confirmed them.
Very nice cover over the fans - it has a very narrow airflow channel. It chokes the fans. After putting it on, you can immediately hear the fans go to high speed - it means that they spin the air in a circle, instead of pumping it through the filter.

The first mod - I cut out the entire inner part of the lid with a weight. The fans are exposed.
I measured the air flow - almost 50% better flow with such a cut cover. And almost 6dB quieter operation (at full 24v power supply)

Because my pliers-cut cover looked ugly - I printed a new one, as in the picture - already with holes cut out for the fans.
[image]
Success.
Quiet (well, maybe not super quiet, but quieter than the original) with higher airflow. As a result - you can set almost half the power supply - there is the same air flow and it is very quiet.

thenickdude, 2023-04-02:
6dB is huge, could you share your STL? I'd love to try it too.

thenickdude, 2023-04-03:
Makes sense, I'll try that too

0ndsk4, 2023-04-10:
Thanks for this. Changing the lid design shortly.

bistory, 2023-05-21:
> 6dB is huge, could you share your STL? I'd love to try it too.

If you want, here is a quickly modded lid (I just cut 38mm holes into the existing lid)
[LID.stl.zip](https://github.com/nevermore3d/Nevermore_Micro/files/11524019/LID.stl.zip)

I'm printing it right now :-)

0ndsk4, 2023-06-02:
Thanks. Testing has shown the lid design wasn't optimal. This "bug" has now been squashed in the new Micro V6. When cad files are online its likely possible to use them to update the V5 lid for people who do not wish to upgrade their micros.

closed 2023-06-02 by 0ndsk4