nevermore3d/Nevermore_Micro #35: [V6] Bridging issue plenum

max-cz, 2023-06-04:
Hi

Is this an errors or how is the printer going to print these? 
Layer 347.

[image]

scheini79, 2023-06-04:
> Hi
> 
> Is this an errors or how is the printer going to print these? Layer 347. Sliced with PIF standard in Orca and Superslicer.
> 
> [image]

You need to turn it 90 degree. You’re printing it in the wrong direction.

eambos, 2023-06-04:
This issue was reported on the Nevermore Discord. It's due to a 0.4 degree misalignment. 
[image]

scheini79, 2023-06-04:
> This issue was reported on the Nevermore Discord. It's due to a 0.4 degree misalignment.
> [image]

But in his picture it’s clearly standing, not laying flat.

0ndsk4, 2023-06-04:
Thank you. As noted, somehow the bridge had a 0.4 degree angle causing it to be sliced in half. It’s been corrected in the cad file, and a new stl export will be up soon!

closed 2023-06-04 by 0ndsk4