nevermore3d/Nevermore_Micro #39: V6 Issues

forReason, 2023-06-09:
I am having trouble printing the catridge and the part holding the fans:
For the catridge:
- The support panel is unprintaple, I resliced it, printed at 30mm/s and the support panel just becomes a bunch of spaghetties.

- The part holding the fans has three primary issues for me: 
    - The fans provided by ldo do not fit perfectly
    - there are no channeld for the cables
    - the top lit does not fit (its loo long)

0ndsk4, 2023-06-09:
Check out the plenum for sunon geometry fans uploaded today. The regular does fit delta, gdstime and winsinn. The new sunon should work for honeybadger, likely ldo uses sunon geometry as it was recommended for v5.

I have not had a single support failure myself, and am interested in what doesn’t work
1) did you try yesterdays cartridge panel with even more attachment points?
2) do you use a nozzle >0.4mm, making single wall 0.4mm extrusions unreliable? If so what size?
3) do you use 0.2layers and 0.2 bottom layers so the integrated support lines up with layer heights? If not, how it prints can be somewhat slicer dependent. This has been prototyped using cura. 

No wire path is a feature, not a flaw:)

johnbarratt, 2023-06-11:
> the top lit does not fit (its loo long)
Seeing this as well as per pic, also is too wide by a few mm:
[image]

This is with the latest files for v6. Also, FWIW, my supports didn't fail.

closed 2023-06-16 by forReason