nevermore3d/Nevermore_Micro #42: v6 - please provide assembly photos

garyd9, 2023-06-19:
A couple of photo's showing the assembly of the v6 would go a long way to making it easier to follow the directions.  While I was able to print and (finally) assemble it, there were a couple of "huh?" moments.  

One issue I never managed to resolve to my satisfaction was where to run the fan wires inside the plenum.  Right now, the plenum cover bulges out a bit due not having anyplace to run wires (or not knowing where they should be run.)

Finally, I'm not thrilled with the plenum's "snap close" lid.  If I ever need to open that back up to service a fan or something, I'm not sure I'd be able to do so without breaking the plastic.  If it's secured with a screw, why ALSO make it snap closed?

0ndsk4, 2023-06-19:
Thanks for the feedback, it’s greatly appreciated!

The wire space is the inside hollowed out surface of the lid - there should be close to 2mm for wires with no bulging as long as the wires 1) don’t overlap, and 2) the outermost fan has the wires removed from the fan casing wire path and folded in the other direction so it doesn’t interfere with the thicker edge/rim of the lid. If you want, you can hold the wires with a piece of tape while securing the lid, but once the lid is on it will hold the wires securely with no tedious wire path routing. 

As for the lid fastening, Im happy for hear input on getting this to work better. . 
Lid is tight and the main “snaps” are located precisely above the fan exhausts to keep an air leak path from forming. The bolt is not really needed, as stated, but was added for high heat applications where plastic wants to deform (v5 lids deformed without a bolt). For the same reason, a single bolt in the center would not keep the edges from warping if exposed to high heat (without edges being held in place, ie by snaps). There was also limited room for a bolt insert in each corner of the lid and still keeping everything symmetrical. 

Those were the main considerations forming the current design. I agree it could be improved, not yet sure on the best way to proceed.

garyd9, 2023-06-20:
Perhaps I should have paid closer attention to the space for the wires.  Well, bulging or not, the v6 appears to be working at least as well as the v5.  I really had no good reason for printing the new version, my v5 has been working for quite a while without issues (and no warping.)

Some other random comments:

I was very impressed with your built-in supports on the models: they worked perfectly.

The sliding cover for the cartridge is clever and installed as described in the GitHub readme.md. 

Thank you for including space for 2x WAGO’s.  That made my life much easier.

I really haven’t noticed any difference in airflow between the v5 and v6, but I really have no way to measure (and it wouldn’t make sense to compare unless I could source identical fans.)

It seems like the v6 holds less carbon than the v5, but that could just be a matter of perception (because of the smaller opening to pour in carbon.)

Again, thank you sharing all your hard work.  It’s appreciated.  (This issue can be closed/resolved.)

closed 2023-06-20 by garyd9