nevermore3d/Nevermore_Micro #43: V6 Cartridge Lid Design

ptsiampas, 2023-06-23:
I love what you have done with the V6; the plenum is very nice and practical and much better than the V5.

I have been trying to fit the Jalousie, it has the flex required but because the rails its supposed to slide in are not perfect it is challenging to fit correctly; it looks great, but for ease of assembly, it's simply too hard for me to fit it as described.  I have printed it according to the settings suggested and followed the install instructions by warming up the part but no luck. I am up to my sixth print of it and still no luck.

Can you add a slid lid like the V5 for the cartridge?

Hotpocketdeath, 2023-06-25:
I concur with this.

I have been struggling to get the Jalousie slide to work, and it is way too difficult.  I've tried different materials and modified the designs a bit from the f3d files to try and put some more tolerance in and I just can't get it to work smoothly.  

I'm burning through filament that I feel like this is still in prototype phase and not a final design.

I think it would be better to get rid of the bend altogether and just have a solid piece that slides straight out the side and held in by a screw.

0ndsk4, 2023-06-26:
Hi there -

Feel your frustration. The jalousie has been tested for 500+ openings, after which too much pain set in, so I know it can and does work well - for me and many others. 

Hope it will become more clear in the coming weeks what filament types and slicer settings that seems to have problems with the jalousie, as this isnt an experience I wish on nevermore users. I'll leave this Issue open for others having problems to chime in.

Regarding a different lid solution - not impossible in due time to best serve users with jalousie woes, and Im happy to recieve mods which Ill upload. But currently theres little time to go back to the drawing board. As for keeping it simple, nothing is simpler (to use!) than a jalousie slider. Hopefully with more information on the matter, it can become as easy to print as to use, and we can start to agree to agree on this :)

Please state your tested filament, slicer, and in what matter the jalousie fails (bad fit, breaking, whatever) and lets see if a pattern emerges!

agravelot, 2023-06-26:
I also had a lot of trouble inserting the blind slider into the cartridge. What I had to do was clean the blind's slide path in the cartridge. 

I strongly suspect this was due to my seams being positioned exactly on track. Perhaps by slightly reducing the width and/or height of the jalousie, we can increase tolerances and be more forgiving of misplaced seams and bad overhangs.

[image]

Adding a small filet on the Jalousie can help reducing grip into seams.

[image]

Printed on OrcaSlicer 1.6.3

emolitor, 2023-07-02:
I ended up solving my woes with the Jalousie door by printing it in PETG at 98% scale. With ABS it was quite hit or miss due to stiffness of eSUN ABS+ (3dqf ABS did work a bit better than eSUN ABS+) whether it would break on first insert. First attempt with PETG was very tight until I realised I needed to compensate the scale a bit due to (I supsect) less shrinkage vs the ABS printed cartridge.

I suspect that the V6 requires a relatively well tuned machine, a somewhat 'pliable' ABS and slicer settings (as noted in the docs) that generate a compliant jalousie door. With very tight tollerances the v6 design isn't very forgiving of any issues with printer tuning, material or slicer. I suspect that my setup is needs more tuning although simply changing material worked quite well for me.

w35t13, 2023-07-02:
> I ended up solving my woes with the Jalousie door by printing it in PETG at 98% scale. With ABS it was quite hit or miss due to stiffness of eSUN ABS+ (3dqf ABS did work a bit better than eSUN ABS+) [...]

I printed mine in esun abs+ numerous times and they all worked. even the earlier alpha's

emolitor, 2023-07-02:
> I printed mine in esun abs+ numerous times and they all worked. even the earlier alpha's

I tried two different rolls of eSUN ABS+ on three printers (V0, Trident and X1) with only one out of six working moderately well. What slicer and what settings are you using? Obviously it works for some people and not for others so curious what is different or, as I said in my comment, if it's just down to tuning parameters. FWIW the version I printed on my Trident was the one usable one I was able to get out of eSUN ABS+.

n0ughtchill, 2023-07-02:
I was having a bit of trouble with Polylight ASA, I just shrunk the width by 1% and it's been perfect. My printer is a bit more dialed in since I printed it though so I'm not even sure if that is necessary.

0ndsk4, 2023-07-02:
Are you using ironing on top layers?

In such a thin piece like only three structural layers it might make a difference, creating an annealing effect. As I never have had a jalousie break on me, I’m trying to come up with reasons as to why.

av8er, 2023-07-03:
I also am having trouble with the door. I've tried the settings suggested with Polymaker ASA and a generic ABS. I think that the tight radius that the door has to make is part of the problem. I've noticed that once the door has the bend in it it causes friction in the straight section. I too would suggest that you make the door straight. there is no need to make the turn around the corner except for the cool factor. causes more problems than it's worth.

0ndsk4, 2023-07-03:
While I'm not opposed to a straight lid mod, the bend has a function besides looks - its what keeps the lid secure. A single axis insertion is only secure as long as its bolted or latched shut, but can otherwise slide out. Two-axis means it will always stay in place regardless of how its handled or if the cartridge is hit and falls from the magnets spilling carbon everywhere. been there, done that.

av8er, 2023-07-03:
Ok, but you could have a feature that you could press down then slide it in for the last few mm to engage the lid. Right now the sharp radius that the door has to make is unsuitable for ABS/ASA it fatigues the material and it will fail. since the "secret sauce" settings that you have for printing elude the rest of us this design is going to cause you more trouble in support than it's cool factor is worth.

emolitor, 2023-07-03:
Good question, I will have to check my slicer settings when I get back home
later in the week. You may be on to something regarding the top layer.

mustang5269, 2023-07-04:
I broke three of them while trying to get past the first bend until I realized that I had it upside down🤦‍♂️ for obvious reason it went right in so make sure your not making the same stupid mistake I did.

av8er, 2023-07-06:
I'm up to about 7 of these things with no success. Tried ABS, ASA & Petg all will fail. Has anyone figured out the settings necessary to get a working one of these?

av8er, 2023-07-17:
I've found a way to get this to work for me. The bottom 3 solid layers of part is 0.6mm thick. On all of my prints this makes it too stiff and brittle to bend easily. What I have done is place the part on the bed and then lower it by 0.2mm,(change Z position from 1.43 to 1.23) this will make the bottom part only 2 layers thick. On my setup that works just fine and I don't see any fatigue cracking on the bottom of the part. I used the recommended settings. just make sure that both bottom layers are running the length of the part (Infill Angle).   

Note: this has not been tested for longevity yet.

sru-tx, 2023-07-17:
I printed the jalousie door two different times.  I applied just a bit too much heat to the end of it as I was inserting it into the plenum and distorted the end a bit. It went in but cracked at the hinge points at multiple places after a couple cycles. I reprinted it, cleaned up the plenum slide areas and the doors slide edges, and then heated and inserted.  Again, after a few cycles the hinge areas of the door all turned white and cracked and split. I'm using Microcenter brand Inland ABS with your Github entry's print suggestions.  Liked the concept but just couldn't get it to work.

0ndsk4, 2023-07-18:
> I've found a way to get this to work for me. The bottom 3 solid layers of part is 0.6mm thick. On all of my prints this makes it too stiff and brittle to bend easily. What I have done is place the part on the bed and then lower it by 0.2mm,(Superslicer: change Z position from 1.43 to 1.23) this will make the bottom part only 2 layers thick. On my setup that works just fine and I don't see any fatigue cracking on the bottom of the part. I used the recommended settings. just make sure that both bottom layers are running the length of the part (Infill Angle).
> 
> Note: this has not been tested for longevity yet.

I uploaded a thin jalousie to replicate this effect. Added 0.2mm thickness on the opposite side by extending the jalousie ridges, so hopefully it will be almost as strong at the edge (too thin and the sliding path wants to break free).

Id be happy if someone who has had issue would want to trial this version, made for stiffer filaments.

0ndsk4, 2023-07-19:
[image]

The new thin jalousie option is significantly more bendable - with my abs+ filament it can be rolled into a 20mm roll, compared to 25mm using the regular version. Inspecting the rolls, the thicker roll also does not have as smooth a curve, suggesting it both bends better and smoother.

RedSinkDirty, 2023-07-22:
Print my jalaussy door in asa, it was kinda stiff to get in but it worked, I had to print 3 doors till I had the settings down to get it to print clean, just put all my speeds to 40mm/s and doubled my fan power and dropped my first layer to 10mm/s from 50.  And the in the main filter body I had to really carefully clean out the track with a razor, the track had defects at the points where the support was touching. its certainly never gonna be a one handed job on mine

emolitor, 2023-07-26:
Printed the updated door in PETG and working well, thank you.

queezz, 2023-08-11:
> I printed the revised door design in the ABS filament that previously failed. The new thinner door design works! thank you!

Same here, `[a]V6_Jalousie_Slicer_THIN[Stiff_filament][7_walls].3mf` slid right in to the case. Used Voron settings + 7 walls, printed in overture abs.

rveda, 2023-08-24:
I printed the door in PrimaSelect ABS with the suggested settings. It slid into place with some force and seemed to work. Hiwever, after a few openings it broke in half. I think I will try to reprint in PETG when I have some spare time. 

I would prefer a snap-in lid or something less finicky that the slide in door.

NoahWL, 2023-09-02:
I printed the "still filament" XL jalousie in Polymaker ASA, thinking I'd have a better chance of it surviving.  I managed to get it all in with great difficulty but getting it around that first turn just required too much force and I pulled it apart.  The flexibility of the jalousie didn't seem to be the problem; the fit was just too tight in that initial path and bend.  I ran some sandpaper through that path but it didn't help much.  I then printed the jalousie at 70% Z scale, but it still required a lot of force to insert.  Finally, I lightly sanded the edges of the jalousie - the thin edges that sit in the track - and it slid right in and made it around the first bend easily.

My XL cartridge definitely warped a bit.  It stayed stuck to the bed but once it was removed (after cooling overnight) the edges of the jalousie opening pulled up toward the top of the model.  With so little rigidity around that opening I think it's fairly prone to warping, though it could just be my printer/settings.

I think making the short "insertion point" slot/track slightly wider could help, as could increasing the radius of that first turn.  On the XL cartridge especially I think increasing that turn radius would be well worth the sacrifice of carbon storage space.

0ndsk4, 2023-09-02:
> 

Thank you for excellent feedback. It seems to be very common for people to pull the jalousie in, instead of pushing it. A good way of destroying it easier - it was made to push in, a mm at a time around the first bend.

Instead of telling everybody theyre wrong, I just uploaded a new thin jalousie with a pull tab. Should make insertion very easy. Any takers?

Hotpocketdeath, 2023-09-03:
So I didn't bother dealing with the Jalousie anymore and just did a redesign and made it a single solid sliding door.  Printed out just fine and the slides right in.  Tolerances came out really good first try.

Tolerances are so good in fact, I think I could remove the screw and just switch it to a detent instead.
[image]

0ndsk4, 2023-09-03:
Looks great, push it as a mod as a folder in the stl directory please :)

Hotpocketdeath, 2023-09-03:
> Looks great, push it as a mod as a folder in the stl directory please :)

I don't have push rights to do that.

0ndsk4, 2023-09-03:
Sorry, sent an invite now I believe. I’m not very GitHub fluent.

Hotpocketdeath, 2023-09-03:
Alright, the model has been uploaded.  Feel free to try it out.

jwyse, 2023-09-08:
@Hotpocketdeath any chance you might make and share the straight cartridge model with the same modification (for Voron V0)?

thunderkeys, 2024-02-16:
STL should be reoriented for proper printing orientation in the github - I believe it is incorrect as it sits today.