nevermore3d/Nevermore_Micro #47: Plenum wall too thin

Matt2916, 2023-07-09:
The wall on the upper right hand of the V6_Plenum is too thin and slices with a tiny hole.
[image]

0ndsk4, 2023-07-09:
Thx. This was noted on discord too. Expect a fix when back from vacationing. It’s possible to slice the wall continuously tweaking the slicer settings, which is why this wasn’t noticed earlier - however, there will always be a surface imperfection from being just a single wall. This will be updated.

Matt2916, 2023-07-09:
My v5 works still fine, so I'll wait for the update - have a great vacation!

0ndsk4, 2023-07-20:
issue has now been fixed in alla files, thanks!

closed 2023-07-20 by 0ndsk4