nevermore3d/Nevermore_Micro #49: the poke yoke tongue on the v6 cartridge doesn't actually stop reversal on the straight cart.

RedSinkDirty, 2023-07-22:
(sorry if this is a overloaded issue Its kind of just misc feedback and soft issues. Feel free to just bonk and close it)

Maybe its not meant to stop the filter from being put on the other way around, but the tongue on the filter that pokes into the main body doesn't do anything on the straight cartridge when flipped around. It looks like I have the filter on backwards by the tongue but this is actually the way I would want the filter for my trident,
[image]

and then for the trident mount am I supposed to put the nevermore on the back wall or the side, a little more instruction on mounting might be nice, just saying it wasn't clear, Ill figure it out or make my own mount. Id guess more mounting pictures/directions are planned. If I had noticed before I printed it I probably would have flipped the text for the way I'm mounting it, lol youre using the v6 Im on 9 delta. I also didn't print the 2020 mount because I didn't read but also because it was labeled optional in the folder. Where's the other heatset insert go?
[image]

and and, I ended up cutting a lot of the grating out of the top of the straight cartridge, the profile(default prusa voron v2) I had printed it with had 3mm mini travel after retract and with my abs not being extremely dry, every hole was pretty much crossed out with stringing. it'll be fine at least with your pellets nothing seem to fall out. Probably would have printed a version with the more open grate anyway since its gonna be standing up in my trident
[image]
or maybe its just that prusa 2.6 wants to do a line of bridges before perimeters but it starts and stops extrusion, and the wall drags plastic out of the nozzle so its not primed to start and stop like it tries to. and the it came back later to finish some of them, I wonder if a small  gap below the first layer of the cross grating would turn it into only 5 continues bridges, maybe I just have a setting funky, I messed around a little but it didn't seem to want to act differently.
[image]

I forgot to take pictures but winsin fans worked fine I just had to trim off two of the closure snaps for them to sit well, the spring clips hold them well.
The 3x jst mount seemed to work well, I plugged in the connections and then cut and resoldered the wires to the fans so there wasn't much extra wire.

RedSinkDirty, 2023-07-22:
doh the prusa screen shot is the other side of grating. eh that side was pretty messy top but better

RedSinkDirty, 2023-07-23:
Is this how the trident mount was intended? And I had to plug the unused mounting hole, soooo much air was coming out of it.
[image]

0ndsk4, 2023-07-23:
I believe the trident mount is for mounting it back+side under the bed at one or both sides of the T-shaped extrusion. 

Intention of the “key” on the cartridge was to ensure perfect attachment when you can’t see the plenum (ie under the bed of a v2). Up side down attachment is not an issue then.

RedSinkDirty, 2023-07-23:
> I believe the trident mount is for mounting it back+side under the bed at one or both sides of the T-shaped extrusion.

I held it up on the t frame of the bed it didn't really seem to make sense anywhere on a 300 size because it doesn't reach past the edge of the bed.  on a 250 I guess that would make sense. and then on the back wall along the floor it cant reach past the z motor mount "plate" and the outer corner extrusion has wires running there.  so the place in the second picture wasn't really a option.  

I'm happy with where I have it now though. I think the air circulation path will be good here.

closed 2023-07-24 by RedSinkDirty