nevermore3d/Nevermore_Micro #53: V6 plenum for V0 with Winsinn fans

jwyse, 2023-09-09:
**My Winsinn blowers don't fit properly in the "Delta_GDStime_Winsinn" plenum, but they fit well in the "Sunon_Honeybadger" plenum.**

I have Winsinn 5015 blowers (just purchased from Amazon US), so I printed the corresponding plenum for my Voron 0, `V6/v0/v0_Plenum_Delta_GDStime_Winsinn.stl`.  The blowers fit very poorly due to the clips that hold the blower shell together and insufficient clearance in the plenum.  The only difference I can see between the two plenum models (I only looked at the v0-specific variations) is the clearance for the blower clips.  I did a quick draft-quality PLA print of the other plenum model (`V6/v0/v0_Plenum_Sunon_Honeybadger.stl`) just out of curiosity, and the Winsinn blowers fit in that one much better.

Has anyone else experienced this?  Maybe the Winsinn product spec changed, or perhaps the Nevermore models/documentation could be improved by mentioning possible variations in the wild.

[image]
[image]
[image]
[image]
[image]
[image]

CRTified, 2024-01-14:
Still bootstrapping ABS/ASA-printing, so I've printed the `V6/v0/v0_Plenum_Delta_GDStime_Winsinn.stl` out of PETG. Can confirm that it's not really a good fit.

What file did you end up using?

[image]

[image]

CRTified, 2024-01-15:
Being a bit more generous while removing the support made it fit for me. Guess I'll give it a shot :)

[image]
[image]