nevermore3d/Nevermore_Micro #59: Unable to print V6 Cartridge built-in support

Lee-Arthur, 2024-01-02:
V6 Cartridge 90 degree:
Prusa Slicer:
Voron V2.4

Prusa Slicer removes the large built-in support for the sliding door presumably because the slicer perceives inadequate attach points for the support.  Modifying the STL to add a random square object to the support object does force Prusa Slicer to print with the support in place, however, the print still fails when printing the support portion because of inadequate attachment between support and structure.

Unable to print object.

btthrt, 2024-02-27:
Had no problems with prusa slicer. Probably something is messed up with your config.

kitkat0981, 2024-08-12:
I’m having the same issue with orca slicer with asa. The panel keeps warpin
[image]