nevermore3d/Nevermore_Micro #71: Please consider this alternative to the jalousi

matthew-humphrey, 2024-07-05:
I really love the improvements in the Nevermore Micro v6 - everything but the jalousi. I have a well-tuned printer and ASA print profile, and so I had no problems printing it and installing. However, the design seemed fragile, so I spent some time opening and closing it a few times to see how robust it was. As expected, it broke.

I totally get the design to have a toolless design. Maybe a snap-fit lid would work?

I also wanted to pass along this design, which, while not toolless, is very robust and works for both the 90 degree and straight cartridges:

https://www.printables.com/model/514532-nevermore-v6-carbon-filter-easy-lid-remix/

Thanks very much for designing this wonderful add-on - I use it in all my printers.

thijstriemstra, 2024-07-11:
thanks for the suggestion, I'm also going for this alternative.

YoungHillionaire, 2024-07-13:
thank you.
i will try printing this once i install my xol toolhead